martin-ger/esp_wifi_repeater #404: contains malware

polararian, 2020-06-17:
why Port 7777/tcp is open on your firmware . we all know that backdoors use this port , so you need to explain about this !!

jackkum, 2020-06-17:
[README](https://github.com/martin-ger/esp_wifi_repeater/blob/590e464dcb673cc922e62f897b91701ec784b166/README.md)
Advanced configuration has to be done via the command line on the console interface. This console is available either via the serial port at 115200 baud or via tcp port 7777 (e.g. "telnet 192.168.4.1 7777" from a connected STA).

Use the following commands for an initial setup:

set ssid your_home_router's_SSID
set password your_home_router's_password
set ap_ssid ESP's_ssid
set ap_password ESP's_password
show (to check the parameters)
save
reset

polararian, 2020-06-17:
alright and how to close this port ? is your project open source ?

jackkum, 2020-06-17:
You can disable it
https://github.com/martin-ger/esp_wifi_repeater/blob/5fdca67c74f66bc5787883de78e9412378c18bba/user/user_config.h#L69

closed 2020-06-17 by polararian

martin-ger, 2020-06-17:
You can also dynamically change or disable this port:

- set config_port portno: sets the port number of the console login (default is 7777, 0 disables remote console config)

polararian, 2020-06-17:
could you explain more about how to disable ports and enable them , and tell me is it possible to change the frontend of config page (192.168.4.1) ?

RibeiroSantos, 2020-06-17:
I suggest you to first read all documentation about the project. 🙂

polararian, 2020-06-17:
i could not find anything about setting password on telnet or ssh , could you tell me the command ?
😉

martin-ger, 2020-06-17:
There is non password other than the "lock" password.

- lock [password]: saves and locks the current config, changes are not allowed. Password can be left open if already set before (Default is the password of the uplink WiFi)
- unlock password: unlocks the config, requires password from the lock command

polararian, 2020-06-17:
i know this but i want to set ssh with username and password

martin-ger, 2020-06-17:
No way. There is no ssh demon for the ESP8266 - at least none that I am aware of.

polararian, 2020-06-17:
thanks for support 💎