martin-ger/esp_wifi_repeater #407: Boot loop since WPA2 credentials entered

gaijinsr, 2020-06-23:
Hi,

since I added WPA2 credentials (the eduroam use case as described in the manual...)
my ESP is stuck in a boot loop, the ouput below is produced in 3-5sec.

Any idea what is going wrong here?

Any chance of recovery short of zeroing the flash?

Best,           Stefan

` ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 1328, room 16 
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8 
tail 4
chksum 0x34
csum 0x34

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz

Booting rom 0.
⸮⸮⸮⸮o⸮r⸮⸮'|⸮$dld`#⸮⸮|;⸮l⸮g⸮⸮'⸮ $`⸮⸮s⸮l⸮l⸮$⸮

WiFi Repeater V2.2.15 starting

running rom 0
Config found and loaded (3216 Bytes)
WPA2 ENTERPRISE VERSION: [v2.0] enable
Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
mode : sta(98:f4:ab:e4:7b:f2) + softAP(9a:f4:ab:e4:7b:f2)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
EAP-TTLS: Start (server ver=0, own ver=0)
E:M 1568
Fatal exception 28(LoadProhibitedCause):
epc1=0x40216923, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000
⸮
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 1328, room 16 
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8 
tail 4
chksum 0x34
csum 0x34

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz

Booting rom 0.
⸮⸮⸮⸮'⸮;⸮⸮g|⸮ddld`c⸮⸮|{⸮l⸮o⸮⸮g⸮ d`⸮⸮s⸮l⸮l⸮d⸮

WiFi Repeater V2.2.15 starting

running rom 0
Config found and loaded (3216 Bytes)
WPA2 ENTERPRISE VERSION: [v2.0] enable
Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
mode : sta(98:f4:ab:e4:7b:f2) + softAP(9a:f4:ab:e4:7b:f2)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
EAP-TTLS: Start (server ver=0, own ver=0)
E:M 1568
Fatal exception 28(LoadProhibitedCause):
epc1=0x40216923, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000
⸮
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 1328, room 16 
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8 
tail 4
chksum 0x34
csum 0x34

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz

Booting rom 0.
⸮⸮⸮⸮g⸮{⸮⸮o|⸮lldl c⸮⸮<{⸮l⸮o⸮⸮g⸮ d`⸮⸮s⸮l⸮l⸮$⸮

WiFi Repeater V2.2.15 starting

running rom 0
Config found and loaded (3216 Bytes)
WPA2 ENTERPRISE VERSION: [v2.0] enable
Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
mode : sta(98:f4:ab:e4:7b:f2) + softAP(9a:f4:ab:e4:7b:f2)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
EAP-TTLS: Start (server ver=0, own ver=0)
E:M 1568
Fatal exception 28(LoadProhibitedCause):
epc1=0x40216923, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000
⸮
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 1328, room 16 
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8 
tail 4
chksum 0x34
csum 0x34

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz

Booting rom 0.
⸮⸮⸮⸮g⸮{⸮⸮o|⸮lldl c⸮⸮<{⸮l⸮o⸮⸮'⸮ $`⸮⸮{⸮d⸮d⸮l⸮

WiFi Repeater V2.2.15 starting

running rom 0
Config found and loaded (3216 Bytes)
WPA2 ENTERPRISE VERSION: [v2.0] enable
Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
mode : sta(98:f4:ab:e4:7b:f2) + softAP(9a:f4:ab:e4:7b:f2)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
E:M 1568
Fatal exception 28(LoadProhibitedCause):
epc1=0x40216923, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000
⸮
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 1328, room 16 
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8 
tail 4
chksum 0x34
csum 0x34

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz

Booting rom 0.
⸮⸮⸮⸮n⸮;⸮⸮g|⸮ddld`c⸮⸮|{⸮l⸮o⸮⸮g⸮ d`⸮⸮s⸮l⸮l⸮l⸮

WiFi Repeater V2.2.15 starting

running rom 0
Config found and loaded (3216 Bytes)
WPA2 ENTERPRISE VERSION: [v2.0] enable
Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
mode : sta(98:f4:ab:e4:7b:f2) + softAP(9a:f4:ab:e4:7b:f2)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
E:M 1568
Fatal exception 28(LoadProhibitedCause):
epc1=0x40216923, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000
⸮
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 1328, room 16 
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8 
tail 4
chksum 0x34
csum 0x34

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz

Booting rom 0.
⸮⸮⸮⸮g⸮{⸮⸮o|⸮lldl c⸮⸮<{⸮l⸮o⸮⸮g⸮ d`⸮⸮r⸮l⸮l⸮d⸮

WiFi Repeater V2.2.15 starting

running rom 0
Config found and loaded (3216 Bytes)
WPA2 ENTERPRISE VERSION: [v2.0] enable
Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
mode : sta(98:f4:ab:e4:7b:f2) + softAP(9a:f4:ab:e4:7b:f2)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
EAP-TTLS: Start (server ver=0, own ver=0)
E:M 1568
Fatal exception 28(LoadProhibitedCause):
epc1=0x40216923, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000
⸮
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
`

martin-ger, 2020-06-23:
"E:M XXX" means out of memory. Guess, the eduroam certificate is too large for the RAM of the ESP8266. Can you check about the size of the certificate (sometimes it contains multiple Sub-CAs and their keys), perhaps on another computer?

No idea, how to work around that...

gaijinsr, 2020-06-24:
Hi Martin, thanks for the quick answer. I captured the authentication process on a different computer and extracted the packet with the certificates,

...
        TLSv1.2 Record Layer: Handshake Protocol: Server Hello
        TLSv1.2 Record Layer: Handshake Protocol: Certificate
            Content Type: Handshake (22)
            Version: TLS 1.2 (0x0303)
            Length: 5327
            Handshake Protocol: Certificate
                Handshake Type: Certificate (11)
                Length: 5323
                Certificates Length: 5320
                Certificates (5320 bytes)
                    Certificate Length: 2553
                    Certificate: 
...

The packet is repeated several times, but it always has 5327 bytes for the three certificates. I know the ESP8266 is short on memory, but 5K should not be a problem, should it?

gaijinsr, 2020-06-25:
I now tried using the ready-made firmware (previously I had compiled it myself),
this solved the out-of-memory problem. Unfortunately, while everything is fine until
I use PEAP, as soon as I try to connect to eduroam, I get constant WDT resets:

...
epeater V2.2.15 starting
running rom 0
Config found and loaded (3216 Bytes)
WPA2 ENTERPRISE VERSION: [v2.0] enable
Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
mode : sta(98:f4:ab:e4:7b:f2) + softAP(9a:f4:ab:e4:7b:f2)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
EAP-TTLS: Start (server ver=0, own ver=0)

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)
wdt reset
load 0x40100000, len 1328, room 16 
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8 
tail 4
chksum 0x34
csum 0x34

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DIO
Flash Speed:  80 MHz
Booting rom 0.

AReubens, 2020-11-21:
I have the exact same issue. I've used a range of ESP-01s and have gone from a simplistic to overkill circuit to make sure power fluctuations weren't causing the problem.

As above, when building from source:
> Fatal exception 28(LoadProhibitedCause):
> epc1=0x40216923, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000

And when flashing the precompiled binaries:
> ets Jan  8 2013,rst cause:4, boot mode:(3,6)

@gaijinsr Did you have any luck?

abskjha, 2021-03-23:
I have got the similar problem with NodeMCU.
@AReubens @gaijinsr Have you got any success with this boot loop?

AReubens, 2021-03-24:
No progress made with it I'm afraid, I've been using a ESP32 instead

abskjha, 2021-03-25:
@AReubens, that is still good. Have you been able to connect to wpa-enterprise network with ESP32? Have you used this source code https://github.com/martin-ger/esp32_nat_router

AReubens, 2021-03-26:
Ah, so this is has been the bane of a project for a while. I've been attempting to connect to a university wpa-enterprise network that has some oddities in the radius server config that I'm not 100% clear on. If you're having this same problem then it may be that the networks are configured similarly...

I've been able to connect to this wpa-enterprise network using both a esp8266 and an esp32 when programmed through the arduino ide using their respective wpa examples from the arduino ide, but neither of the include the NAT functionality I wanted this project for since the core/idf versions are older. The newer esp32 idf does include NAT support and that's what the the esp32 project you linked requires, however when using the newer version of the esp-idf which supports NAT, I've been unable to connect to the wpa-enterpirse network.

In sort, currently I've been able to have either NAT support or WPA-enterprise connection success, but not both. And that's on both esp8266 and esp32 devices.

Blitzbirnep, 2021-07-11:
Hey guys, 

any progress on this issue? I have the same problem as reported above.

First I have set up the ssid and peap credentials, saved the config and restarted the esp8266. 
However afterward it is stuck in a boot loop.

I managed to connect via a Linux computer with the following wifi settings: 
- WPA 2 Enterprise
- TTLS 
- Anonymous identity empty
- No CA certificate is required (checked)
- mschapv2 (no eap) 
- username
- password

I tried two different power supplies (laptop, external 2.5 A supply) same effects.

The terminal log is below:

martin-ger, 2021-07-11:
Think, this happens when the certificate provided is too large for the memory. No idea, how to fix that...

bear-zd, 2022-12-04:
if this problem solved?