martin-ger/esp_wifi_repeater #410: no restart after firmware download

therwastl07, 2020-08-13:
Hello!

I appreciate your effort and tried to use your system because the auto-mesh is rather unique.
I tried to download the latest onto a 12-E and a 12-F NodeMCU and in both cases they did not start to show the AP.
With the 12-E (for example), I was able to dowload the firmware with the "download tool", after restart and with putty and 74880 baud (only speed without garbage) I read after the reboot a line like:

ets Jan 8 2013,rst  cause:2 , boot mode:(3,6)
ets_main.c

Does anyone has an idea - would be jolly good!!
Thanks - TW7

imayoda, 2020-09-28:
some good hints that saved me a lot of times with 8266:
- ensure good power supply (some boards have flaky regulators or usb connectors)
- erasing all flash contents can bring a solution sometimes
- ensure the right flash reading mode (DIO is usually accepted, but slower solutions like DOUT are present on some manufacturer boards like sonoff basic etc.).

btw if program succesfully starts, it changes baud to 115200. :)

mikedarzen, 2020-11-02:
I saw the same error message.  It seems the problem is in how you download the firmware.   Just right clicking the file (in Windows) didn't work for me.  I suspect it was transferred in a non binary format.   I downloaded the zip file of the repo and flashed from those files and it the error message went away and it booted as expected.

imayoda, 2020-11-19:
@mikedarzen you pointed it right, downloading .zip is always the good thing to do, at least when you are browsing the web github.. :)

imayoda, 2020-11-19:
would be nice to hear news from OP so we can close this issue somehow :)

BoffinTecs, 2020-11-21:
Me I have the esp-01s. I managed to flash the firmware successful but when I restart it in non-programming mode, nothing seems to be happening, I can't see the default AP and when I send commands I don't receive any response.
Thanks in advance.

therwastl07, 2020-12-13:
Sorry for the delay.
I got it running with a capacitor on the power pin.
Grrrr…

closed 2020-12-13 by martin-ger