martin-ger/esp_wifi_repeater #411: Automesh

Bersaker, 2020-08-30:
Hi, have a question.  If I have set the repeater to automesh mode, how can I deactivate it again?
Is that only possible with a hw reset?

martin-ger, 2020-08-30:
`set automesh 0`
on the cli does the job.

Bersaker, 2020-08-31:
Hi. Was ist mit cli gemeint? Hatte vorher den repeater über telnet Port 7777 konfiguriert. Aber bekomme im Automesh kein Zugriff. Habe jetzt neu geflasht. 

Habe aber noch ein anderes Problem. Nutze eine FritzBox und habe zwei Wemos als repeater dazwischen um die Distanz bis zum Garten zu erreichen.

Vorher hatte ich einen xiaomi repeater dazwischen und konnte alle Geräte im selben Netz erreichen (192.168.188.xxx)

Die Box hat die IP 192.168.188.1 der erste wemos hat 10.24.1.1 der zweite hat 10.24.2.1...

Jetzt kann ich aber die Geräte nicht mehr mit 192.168.188.xxx erreichen. 

Arbeitet der xiaomi repeater anders ? 

Wenn ich jetzt auf meinem solar tracker schaue habe ich die IP 010.024.001.003 welche aus dem 192.168.188.xxx Netz nicht erreichbar ist.

martin-ger, 2020-08-31:
CLI meint command line Interface, also die serielle Konsole oder über Port 
7777.

Ja, das Programm arbeitet anders. Genau genommen ist es ja gar kein 
Repeater, sondern ein NAT Router. Das heißt, um rückwärts Konnektivität zu 
erhalten, braucht man einen eingestellten port-forward.

Bersaker, 2020-08-31:
Also kann ich die outdoor Geräte nur über das subnetz erreichen? Wie stelle ich den Port forward ein ?

Bersaker, 2020-08-31:
Was ich meine... der solar tracker , der esp8266 mit tasmota, ein logger für den solar laderegler sollen weiterhin nur im internen Netzwerk miteinander kommunizieren können. Also das ich aus dem 192.168.188.xxx Netzwerk ins 10.24.1.1 bzw. 10.24.2.1 komme und die Daten vom logger ins 192.168.188.xxx Netzwerk Daten senden können.
Ist es ohne weiteres möglich das die Geräte im internen Netzwerk miteinander kommunizieren können ?

martin-ger, 2020-08-31:
Geräte können aus dem ESP-Netz heraus alle Adressen erreichen, nur umgekehrt bräuchte man den Port Forward mit:

`portmap add [TCP|UDP] external_port internal_ip internal_port: adds a port forwarding`

Bersaker, 2020-08-31:
Ok. Ich brauche ja nur die Möglichkeit, das alle Geräte im internen Netzwerk miteinander kommunizieren können.

Bersaker, 2020-09-02:
Bekomme das nicht hin das intern alle Netz miteinander intern kommunizieren können...

CMD>show stats
System uptime: 0:00:40
44 KiB in (338 packets)
83 KiB out (293 packets)
Power supply: 2.890 V
Phy mode: n
Free mem: 48920
STA IP: 192.168.188.62/24 GW: 192.168.188.1
STA RSSI: -65
1 Station connected to SoftAP
Station: f6:c0:52:08:ff:21 - 192.168.4.2
CMD>show route
Routing table:
Network              Dest
192.168.4.0/24       ew1
192.168.188.0/24     ew0
127.0.0.0/8          lo0
default              192.168.188.1

Habe in der Beschreibung was von add route gelesen.

Habe ausprobiert...
 Route add 192.168.4.0/24 192.168.188.62
Oder
Route add 192.168.4 0/24 192.168.188.1

Bei beiden bleibt er im terminal hängen. Bis zum reset.

Bersaker, 2020-09-11:
Der esp8266 nat hat derzeit folgende konfig.

In der fritzbox hat er die IP 192 192.168.188.33

Das subnetz ist jetzt 192.168.4.1
Der logger eines esp32 hat eine weboberfläche
Demnach müsste ich folgendes anwenden...

portmap add TCP 80 192.168.4.2 80

Also wäre der logger jetzt über 192.168.188.33:80 zu erreichen. 
Was mache ich denn wenn ich noch andere Geräte in dem subnetz sind ? Habe ich jetzt nur diese eine Möglichkeit diese über einen anderen Port zu erreichen?

device2000, 2022-02-13:
> Geräte können aus dem ESP-Netz heraus alle Adressen erreichen, nur umgekehrt bräuchte man den Port Forward mit:
> 
> `portmap add [TCP|UDP] external_port internal_ip internal_port: adds a port forwarding`

Hallo Martin,

ich greife diesen Post nochmal auf, weil ich es nicht richtig verstehe und hoffe Du kannst es mir erklären:

Ich habe einen ESP-Wifi-Repeater mit Automesh laufen, funktioniert auch tadellos. Es ist nur ein zweiter ESP der meinen Zähler liest in dem Subnetz.
Habe in meinem HomeRouter einen Routing von der Wfi-Repeater IP 192.168.168.142 zum Netzwerk 10.24.1.0 und kann auf die IP 10.24.1.1 pingen und per Browser zugreifen (ist der Repeater) das Gerät in dem Netz hat 10.24.1.2 das kann ich nicht mal anpingen. Das Gerät sendet tadellos ins Netzwerk.
Ich würde da auch gern per Browser zugreifen, muss ich eine Route einrichten? 

ESP-Repeater 192.168.178.142
Subnnetz 10.24.1.1/24
Gerät 10.24.1.2

Und nun???

martin-ger, 2022-02-13:
`portmap add tcp 8080 10.24.1.2 80`
Dann sollte man mit http://192.168.178.142:8080 auf das Gerät zugreifen können.

device2000, 2022-02-13:
> `portmap add tcp 8080 10.24.1.2 80` Dann sollte man mit http://192.168.178.142:8080 auf das Gerät zugreifen können.

Mit Port 8888 hat es funktioniert! Port 8080 ging nicht = Fehlermeldung "Invalid Parameter"
Jetzt gehts und ich kann prima darauf zugreifen. 
DANKE!

Warum die Route (Routing) nicht reicht verstehe ich dennoch nicht.