martin-ger/esp_wifi_repeater #416: Router configuration

antjnm, 2020-09-25:
I have this problem, with an old Wi-Fi repeater I displayed the ip of connected devices on it, even on the router configured with ip 192.168.54.1 while with the nodemcu I can not. I ask you how to configuarate the esp so that if I am connected to the router I can access the connected devices on the nodemcu.

imayoda, 2020-09-28:
if I understand it right, you need to get in touch with the "routing" command set from CLI.. essentially you have to create routes from the main router to the LAN behind the esp8266 :)

antjnm, 2020-09-28:
thank you very much for the reply, so I have to enable the command "route add network gw 192.168.54.1"?

imayoda, 2020-10-11:
route add network gw: adds a static route to a network (network given CIDR notation ('x.x.x.x/n')) via gateway gw

don't forget CIDR notation
:)