martin-ger/esp_wifi_repeater #417: Disabling transmission to the STA

merlin475, 2020-09-27:
Hi
Is there a way to disable only transmission to the STA? Imagine the ESP is connected to a modem which has access to internet and there is a phone connected to the ESP board. Is there way to only disable transmitting the received data from the phone to the modem yet letting the phone receive data from the modem?
Thank you Martin

martin-ger, 2020-09-27:
You can do this using the ACLs, but prohibiting any packet from the phone to the internet won't work as you might expect: all TCP connections (including HTTP and MQTT) require bi-directional traffic (for connection establishment and ACKs later), even if data is only transmitted in one direction.

merlin475, 2020-09-27:
Thank you for replying so fast. Really appreciate it.
What about set upstream_kbps bitrate and set downstream_kbps bitrate?
If my understanding is correct these are basically the send and receive speed of esp. Can we use an extremely low value to almost stop the transmission in one direction?

martin-ger, 2020-09-28:
Never tested that for that purpose, but yes, that could be an option.