martin-ger/esp_wifi_repeater #418: Can't seem to get to mesh level 2 using Automesh

relhawary, 2020-10-18:
I am using automesh to connect 4 ESP similar to the configuration you have in the picture but I cannot get one of my nodes to be mesh level 2 (they only connect to main AP and do not connect to each other)

Thanks

martin-ger, 2020-10-18:
Try to place them in a row. Probably they all "see" the main AP and it is the shortest link.

relhawary, 2020-10-18:
I tried this configuration but I still didnt get the to mesh level 2
The distance between the main AP and the first repeater node was 2 meters and the second repeater node was 5 meters from 1st repeater (7m from AP main)

martin-ger, 2020-10-18:
This is too close - as long as the second ESP still sees the main AP with reasonable signal strength, it will choose it.

relhawary, 2020-10-18:
How far away should I place the second repeater to get the mesh level 2 appear in the mac address? Do you know the max practical range of esp in meters? Thank you for your prompt reply :D

relhawary, 2020-10-18:
Assume the ESP has 3 nodes connected to it. If I assign a static IP to one of the nodes, will the DHCP adjust accordingly? (meaning if I assign 192.168.1.3 to one, the dhcp will assign to the remaining nodes anything but the IP ending in 3)

martin-ger, 2020-10-18:
Try 10m+.

No DHCP will not recognize static IPs. If you want to use static IP, use a high arddess in the /24 network (250-254), these will not be used ba DHCP.

relhawary, 2020-10-24:
Hello martin
I am building an ESP based project using your firmware. It is a wireless sensor network as you see in the diagram
every AP is an ESP with NAT firmware 
I am trying to put static IP on all nodes so I can ping it from remote server, however the dhcp cannot be overwritten. How would you suggest we ping node1 other than using the IP?

[image]

martin-ger, 2020-10-24:
Due to NAT you cannot ping from the outside (e.g. "remote server") to let's say "node1". NAT hides internal adresses and ports. It is the same for local PCs behind a "normal" NAT home router. For single nodes and services you might define a port forward, but this is not a solution for many nodes with the same services.

What kind of protocol you want to use. If it is MQTT, each node can easily connect to a broker on "remote server" and then also receive commands via subscribed topics. Also HTTP requests from the nodes to the "remote server" work without problems.