martin-ger/esp_wifi_repeater #420: Can't send commands to chip

ItzLarz, 2020-11-08:
I flashed my Wemos D1 Mini with Flash Download Tool. I got a new network at my Wifi settings. I can connect to the network and make changes via 192.168.4.1, but I can't send commands to the chip via Serial Monitor in the Arduino IDE. I know it's the right COM, because if I join or leave the MyAP network i get this message: "station: [mac address of my smarthphone] join, AID = 1". If i just send "help" or another command, nothing happens. Am I doing something wrong?

martin-ger, 2020-11-08:
Check your serial - should work. Might try via telnet 192.168.4.1 7777 should work as well.

ItzLarz, 2020-11-09:
I think I know what the problem is, I just downloaded Mobile Telnet on my phone and it worked! I think it's because of my Norton Firewall and that it doesn't let my pc connect with an open network.

ItzLarz, 2020-11-09:
Thank you for the help!

closed 2020-11-09 by ItzLarz