martin-ger/esp_wifi_repeater #423: Allow internet acces nor local IPs

jeanjeanG, 2020-11-15:
Hi Martin,

You have done such a great job, congratulation and thank you.
When you see the average price for such a device, you are making us saving money! (and have fun doing it)

I'd like to set up a repeater for my friends to **allow them access internet in my place without allowing them access devices on the local network** (IOT stuff).

I have tried several acl configurations including those in your well written documentation without any good result. It seems I don't know how a simple internet access works ^^

I feel like many of us would love this feature and you probably can add it  directly to the documentation. 

Many thanks for your help !!!

martin-ger, 2020-11-15:
Try, where xx.xx.xx.xx is the address range of your home network:

```
    acl from_sta clear
    acl from_sta IP any 255.255.255.255 allow
    acl from_sta UDP any any any 53 allow
    acl from_sta IP any xx.xx.xx.xx/24 deny
    acl from_sta IP any any allow
```

jeanjeanG, 2020-11-15:
**WONDERFUL !!! It works !**
Thank you so much Martin! And you were so fast to reply!

I'm sure it will interest many people.

Have a good day/night! 
May I tip you some how?

martin-ger, 2020-11-15:
Add a section in the README.

closed 2020-11-15 by martin-ger