martin-ger/esp_wifi_repeater #424: Errors during compiling (iram1_0_seg)

teicors, 2020-11-19:
Hi,
I downloaded the code, but during the compiling I get this error:

CC driver/spi.c
CC driver/new_uart.c
CC user/config_flash.c
CC user/ringbuf.c
CC user/rboot-ota.c
CC user/acl.c
CC user/user_main.c
CC user/rboot-api.c
CC user/sys_time.c
CC mqtt/mqtt.c
CC mqtt/mqtt_msg.c
CC mqtt/queue.c
CC mqtt/proto.c
CC mqtt/ringbuf_mqtt.c
CC mqtt/utils.c
CC easygpio/easygpio.c
AR build/app_app.a
/opt/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: build/app.out section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
make: *** [Makefile:133: firmware/0x02000.bin] Errore 1

It seems not enough ram/space: how can I solve this issue ? or may be better: which version of sdk should I use ? I'm using the 2.1.0-18 and I installed the esp-open-lwip-nat too.

I'm trying to recompile the fw in order to hide the passwords from the web page :)

Cheers,
Angelo

martin-ger, 2020-11-19:
Did you enable more compile options? Try to disable those you probably don't need.
Did you enlarge the the webpage? Try to reduce that.

teicors, 2020-11-19:
Thanks for the fastest answer, but I did nothing.
just git clone and make (after changing the BUILD_AREA inside the Makefile in order to point my environment): nothing more !

martin-ger, 2020-11-19:
Did you use the recommended version of the open_sdk? Different versions have different amount of mem left.

teicors, 2020-11-19:
no luck.
I reinstalled the 2.2 sdk following your suggestion, but always the same error:
/opt/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: build/app.out section .text' will not fit in region iram1_0_seg'
just to know, looking at build dir the file app_app.a is 785674 bytes.

martin-ger, 2020-11-19:
Hmm,...

Try, just for testing, to disable some mem conuming feature, e,g,:

 #define		MQTT_CLIENT 0

in user_config.h

teicors, 2020-11-19:
as reported in the build instruction, I tested the "blink" example.. obviously !
I removed the mqtt, web config and WPA2_PEAP. The file in the build is about 699K, but always the error !
#define		WEB_CONFIG 0
#define		MQTT_CLIENT 0
#define		WPA2_PEAP 0
the esp is a 4Mb, do I have to inform/force the linker or not ?

Just to clarify, I'm using /opt/esp-open-sdk as directory for the esp-sdk and I'm using ~/Software/esp8266/project/esp_wifi_repeater as source dir.

martin-ger, 2020-11-19:
This error happens, when there is not enough instruction RAM (think 48 KB or so) available to hold all the required code. It is not directly related to the flash size but to those functions in the code that are not marked as "ICACHE_FLASH_ATTR". Functions that are marked this way are only cached in iRAM and be be reloaded from flash as required. Typically IRQs are not ICACHE_FLASH_ATTR. Sometimes, when I added new functions and forgot the ICACHE_FLASH_ATTR keyword, I got this error.

But now I double checked my build environment and it seems to be fine for me - no such error with the current repo.

teicors, 2020-11-19:
My fault !!
I'm using Sming to develop my projects and the removing from env (unset) SDK_BASE fixed the issue. It was really strange effectively: a full reinstall of software always with the same error.
Thanks for your support and patience !

Cheers,
Angelo

closed 2020-11-19 by teicors