martin-ger/esp_wifi_repeater #426: No routing from main network to client connected to WiFi repeater

pfugazzi, 2020-12-03:
Hi Martin,

I've a problem to connect device from main network to a client connected via WiFi repeater.

The main network is a Raspbery Pi  
      network 192.168.10.0/24
The ESP as WiFi repeater is configured as static IP:
      STA: IP: 192.168.10.100 Netmask: 255.255.255.0 Gateway: 192.168.10.1
      IP:192.168.4.0/24 [NAT]
I've a ESP32CAM connected  to WiFi repeater with a static IP 192.168.4.50

RaspPi (192.168.10.1) < (192.168.10.100) WiFi repeater (192.168.4.1) > (192.168.4.50) ESPCAM

On Raspberry I set a route to network behind WiFi repeater
     sudo ip route add 192.168.4.0/24 via 192.168.10.100

From Rasperry or a PC connected to I'm able to ping the ESPCAM but 'curl' as same as via browser I cannot reach ESPCAM and straming .

While if connect the PC directly to WiFi repeater  it's working.

Seems some settings are missing on WiFi repeater to allow device from main network to access to device on WiFi repeater via TCP/UDP/HTML.

I take a look on readme and issues but I'm not able to address the topic, could you help me?

Thanks in advance
Paolo

I'

martin-ger, 2020-12-03:
By default, the esp_wifi_repeater is a NAT router. Similar to a standard home router you will neet port-forwarding to access nodes and ports in the NATed network (behind the AP of the ESP). You can use the portmap command to do so: https://github.com/martin-ger/esp_wifi_repeater#routing

Another option might be to disable NATing ("set nat 0") and configure static routes on all nodes.

pfugazzi, 2020-12-03:
Martin,
thanks a lot, just a bit more could you please send me an example about how to configure the route?
I'm not expert on the matter and any attempt I did failed.
Thanks in advanceRegardspaolo

  ----Messaggio originale----

 
 Data: 3-dic-2020 12.11
 
 A: "martin-ger/esp_wifi_repeater"<esp_wifi_repeater@noreply.github.com>
 
 Cc: "pfugazzi"<paolo.fugazzi@alice.it>, "Author"<author@noreply.github.com>

martin-ger, 2020-12-03:
If you go for the second option "to disable NATing ("set nat 0") and configure static routes on all nodes." you should do the following:

On the esp (after configuring the WiFi) enter:
```
set net 0
save
```
on the raspberry:
`sudo add route -net 192.168.4.0/24 gw 192.168.10.100`
(as you already did).

Now you can connect to 192.168.4.50 from the raspberry, given that you have configured 192.168.4.1 as default gateway on the espcam.

pfugazzi, 2020-12-03:
Many thanks Martin

  ----Messaggio originale----

 
 Data: 3-dic-2020 20.18