martin-ger/esp_wifi_repeater #427: Router AP ist nicht  x.x.x.1

ningyichensha, 2020-12-05:
Hallo, I like to extend the WLAN of ma car.

But the IP address of the router in the car is: 192.168.8.46. This I cannot change.

The ESP connects to the car and everything look fine, but the app does not get data.

Because in readme says:  router is always x.x.x.1 I guess this could be the problem.

martin-ger, 2020-12-06:
Can you give me some more insight in the structure of the network?
The car is the AP? Does it give Internet access to the phone?

ningyichensha, 2020-12-06:
Hello Martin, 
I think I just need an extension of the weak wlan, my car provides.
The cars wlan has an AP and my smartphone connects to it. There is no internet involved. Please see the 3 screenshots of the regular functionallity.

[image]
[image]
[image]

When I use the 8266 between car and phone, the 8266 connects to the car and the phone connects to the 8266.
But the app does not talk to the car.

Thanks for your effort.
Peter

martin-ger, 2020-12-06:
Guess, the problem is, that the phone has no route to 192.168.8.0/24, as it is connected to 192.168.4.0/24 AND to the internet via 4G. Default route will be 4G and only connections to 192.168.4.0/24 directly will use WiFi (does it complain about no Internet via WiFi?) all other networks will be searched behind the default route. Does the behavior channge, when you disable 4G data?

ningyichensha, 2020-12-06:
4G is off. Just WLAN is on.
I think your guess is correct. There is no route in the ESP from 192.168.8.0/24 to 192.168.4.0/24.
Maybe the solution is a simple pass through of all packages being sent by car <--> esp8266 <--> phone.

martin-ger, 2020-12-06:
Pass through is not possible with the ESP, so this is not a solution here.

Can you send a ping to 192.168.8.47?
could you try "set dns 192.168.8.47"?

ningyichensha, 2020-12-06:
The sreenshots above (11:04h) are showing the network situation without the esp.  
Just car <--> phone, which works.

The following is the situation  car <--> esp8266 <--> phone.

there is no response to ping on 192.168.8.47.
also after setting dns to 192.168.8.47. 

because the AP in the car is 192.168.8.46, I tried a ping setting dns to 192.168.8.46, which also failed.

CMD>set dns 192.168.8.47
DNS set to 192.168.8.47
CMD>save
Config saved
CMD>ping 192.168.8.47
CMD>ping failed
ping failed
ping failed
ping failed
ping finished (0/4)
CMD>show
Version V2.2.15 (build: Sun Feb 9 22:50:34 2020)
STA: SSID:REMOTE66bdec PW:xxx
DNS: 192.168.8.47
AP: SSID:MyAP PW:none [open] IP:192.168.4.1/24 [NAT]
STA MAC: 48:3f:da:57:df:d2 AP MAC: 4a:3f:da:57:df:d2
STA hostname: ESP_57dfd2
Network console access on port 7777 (mode 3)
Clock speed: 80
MQTT: disabled
CMD>

RibeiroSantos, 2020-12-06:
Do you face same issue with mobile data disconnected?

ningyichensha, 2020-12-06:
mobile data is not involved. The car just communicates via wifi / wlan.

So far it does not respond to a ping, even the esp8266 is connected via STA: SSID:REMOTE66bdec PW:xxx.

martin-ger, 2020-12-06:
mobile data on the phone is important - is it swichted off?

ningyichensha, 2020-12-07:
wifi is on, 4G is off

The test with ping and dns was just between ESP and the car.

You also wrote: Pass through is not possible with the ESP, so this is not a solution here.
Do you mean with your software package, or not possible at all?

martin-ger, 2020-12-07:
In my software it is not possible and I am not aware that anybody was able to implement a bridge like this with Espressif's firmware.

ningyichensha, 2020-12-07:
do you have any idea, why ping from esp to car didn't work?

should I add a route, or does the car's router uses another port?

RibeiroSantos, 2020-12-07:
By curiosity, what is the app you are using?

ningyichensha, 2020-12-07:
The app belongs to the car and is available in the app store. It allows preheating etc.
It just works with wifi. Because of the weak wifi of the car, I need to be very close to the car to activate preheating.
I try to expand the distance with the esp (+ improved antenna), which works on DC.

[image]

RibeiroSantos, 2020-12-07:
Could you please try https://phevwatchdog.net and check if you face same
issue?

ningyichensha, 2020-12-07:
I know watchdog, but it works with OBD2, not with wifi.
As far as I know, it just monitors the car, but can't start preheating etc.

RibeiroSantos, 2020-12-07:
Ah...ok,sorry.
Meanwhile I found this project that maybe could help to understand the
connections needed.
Hope this help.

https://github.com/vzakharchenko/remote-ctrl-gsm

ningyichensha, 2020-12-07:
Thanx for the link. 
That approach is pretty complex. I found no hint concerning the current problem.

I think the ping to the car has to work first.

Because the ESP connected to the cars wifi, a ping should work.
should I add a route, or does the car's router uses another port?

STA: SSID:REMOTE66bdec PW:xxx
DNS: 192.168.8.47
AP: SSID:MyAP PW:none [open] IP:192.168.4.1/24 [NAT]

ningyichensha, 2020-12-07:
Here comes an update.

ping is working!

I tried before telnet with Windows10.
Now I tried with ubuntu and it worked on 192.168.8.46, which is the router of the car.

ningyichensha, 2020-12-08:
Hello Martin,
in the meantime I got ping working after setting dns to the router of the car.
E.g. dns 192.168.8.46   car <---> ESP

Then connecting ESP <---> phone/app:
wifi in the phone (myAP) is: 
router: 192.168.4.1
IP 192.168.4.2 

But the app is still not working with the car
car <---> ESP <---> phone/app

Do you have an idea?

ningyichensha, 2020-12-17:
no solution to my problem
Thanks to your effort

closed 2020-12-17 by ningyichensha