martin-ger/esp_wifi_repeater #430: UDP packets outgoing from AP

alessandrocescon, 2020-12-15:
I have a tello quadcopter connected as AP whith IP 192.168.10.1 which act as DHCP a set first client to 192.168.10.2. 
Tello :
- receive command UDP pachets on port 8889 IP 192.168.10.1 and reply OK or error;
- send telemetry UDP packets on port 8890 IP 0.0.0.0 to connected client 192.168.10.2 (instructions says to create a server on client listening 0.0.0.0:8890);
- sends video stream UDP packets to on port 11111 to connected client 192.168.10.2  (instructions says to create a server on client listening 0.0.0.0:8890);

Now I want add an ESP8266 between the drone and the phone acting as wifirepeater

With the standard configuration my client has IP 192.168.4.2 (esp station has 192.168.4.1).

192.168.4.2<---->ESP STA 192.168.4.1 <---> ESP AP CLI 192.168.10.2 <---> TELLO DRONE 192.168.10.1

I am able to send commands from client 192.168.4.2 to the drone 192.168.10.1 at port 8889 but teletry and streaming packets packet sent from drone to the client (phone) seems don't reach the client. Is the any configuration to **trasparently forward packets sent from drone to phone**?

reiluke, 2020-12-27:
same here, i want to extend my wifi with tello drone, i can control the drone but video is not working

UmutAlihan, 2023-04-24:
same here any solution would be much appreciated