martin-ger/esp_wifi_repeater #435: i can not reset esp!

feredy2011, 2021-01-14:
hi guys.
first, sorry for weak in english.
i have a d1mini . mode is automesh. but now i want change ssid and password on my router, but d1mini dont reset.
for reset i do GPIO0 connect to GND for 3seconds.but not any happen.what is wrong?

martin-ger, 2021-01-14:
The D1 mini has a dedicated reset button. Power off/on will always do the trick...

feredy2011, 2021-01-15:
@martin-ger thank you for help .
But i did not understand. I push reset button for 3 seconds but dont reset.also power off / on and reset 3second again but dont reset. Please guide me step by step for reset factory ( remove old ssid and password )and set a new ssid and password and again automesh...
I weak in english and beginner in d1mini. Please guide step by step . Thank you

martin-ger, 2021-01-15:
Type "reset factory" on the CLI (serial 115200 bps)

feredy2011, 2021-01-15:
Type "reset factory" in terminal ? I m using arduino terminal (serial manitor) 
So just with connect to computer with usb serial , i can reset d1? I think can with gpio reset and again see ssid and again conncet to new router with new ssid and password.

martin-ger, 2021-01-15:
Yes, if the esp_wifi_repeater software is running, you can connect via the Arduino serial terminal.

Generally, I understand with the term "reset" you don't mean a CPU reset, but flashing the esp to some initial state.

If the esp_wifi_repeater software is not running, you can "reset" an ESP (esp. a Wemos Mini) simply by flashing a new arduino program on it.

feredy2011, 2021-01-15:
Reset factory dont work and any happen
[image]

martin-ger, 2021-01-15:
Please try to flash an Arduino sample program. If this works, try to flash esp_wifi_repeater again.

krystian-hebel, 2021-04-03:
I hit the same problem after a typo in SSID. Here is what I did:

```
./esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash -ff 40m -fm dio -fs 32m 0x68000 <any file with non-0 size>
```

The address comes from FLASH_BLOCK_NO: https://github.com/martin-ger/esp_wifi_repeater/blob/master/user/config_flash.h#L17. As a file I used `firmware/0x00000.bin` just because I'm lazy, but any one should do, as long as it doesn't start with a magic number specified just after FLASH_BLOCK_NO.