martin-ger/esp_wifi_repeater #436: In auto mesh mode, where do the clients get their IP addresses from ?

steynaj, 2021-01-20:
Hi Martin,
This is not clear from your description of the auto mesh mode. In auto mesh mode, where do the clients get their IP addresses from ? Is it from the main router or from the NAT routers ?

Thanks
Andries

martin-ger, 2021-01-20:
Locally from the NAT routers from their local subnets. You cannot access these adresses from the outside (as usual with NAT).

steynaj, 2021-01-20:
Martin,
Thanks for replying. Where do you configure the NAT local subnets ?

Andries

martin-ger, 2021-01-20:
This is done automatically "...  Also the IP address of the internal network is adjusted to the mesh level: 10.24.m.0." https://github.com/martin-ger/esp_wifi_repeater#automesh-mode

There is no need to cofigure this - or do you have any issues with the addresses?

steynaj, 2021-01-20:
Martin,
Again thanks for your time. In your example of the auto mesh the IP addresses will be in the range 10.24.1.X or 10.24.2.X is this correct ? If so are the addresses given out by DHCP ?

Andries

martin-ger, 2021-01-20:
Yes, each ESP give out its local 10.24.m.x addresses. ESPs on the same mesh level will use the same addresses in their local nets, but due to NAT, this is no problem.

steynaj, 2021-01-20:
Martin,
Thanks for replying. Is there a way to reserve these addresses in DHCP ? and is there a way to see which client got which IP address ?

Andries

Andries

martin-ger, 2021-01-20:
Locally you can see the assignment on the serial CLI with "show status". Remote login via port 7777 is not so easy in a multilevel mesh setup, as you cannot reach the ESPs directly from the toplevel network (because it is hidden behind NAT). It could be done with a central MQTT broker and the "command_topic" and the response topic of the ESPs, but this is probably an overdoing for just finding out addresses.

With "save dhcp" you should be able to save the current assignment, but this works somewhat unreliably. 

I still don't know what you actually want to achive.

steynaj, 2021-01-20:
Martin,
Actually I want to test the "repeater" capabilities with the auto mesh feature. I have a few ESP-01's that are outside in the garage and want to extend that network to the motorized front gate for presence detection. Using my cell phone to logon to a network at the gate and then automatically open the gate.

Andries.

martin-ger, 2021-01-20:
Guess, the easiest way would be, to connect it upstream via the ESPs to an MQTT broker and control it via MQTT topics.

martin-ger, 2021-10-17:
It is a completely independent approach.

KHUSAINIBINAJ, 2021-11-14:
Pon fiber laser