martin-ger/esp_wifi_repeater #441: How to enter config mode when automesh is used

michielvtb, 2021-03-14:
I searched the issues but could not find it. 
I am using a Wemos D1 mini and after flashing it works fine as an AP. But I want to use the same SSID and password for the AP so I checked automesh. It seems to work fine as I see with Putty over the serial interface. But how do I get in configuration mode again? I want to use a different SSID in the future, and now connecting to  192.168.4.1 does not work anymore afterI have saved automesh mode.
As a workaround I erased the flash, and flashed again, but is this really necessary?

martin-ger, 2021-03-14:
Either use the serial CLI or try to connct via WiFi and goto 10.24.1.1.

michielvtb, 2021-03-14:
10.24.1.1 works, thank you for your quick reply!

closed 2021-03-14 by michielvtb