martin-ger/esp_wifi_repeater #442: ENC28j60 No Internet

slowpoison4, 2021-03-15:
Hi
I have a NODEMCU ESP8266 and an ENC28J60 Ethernet Module.

I've been able to successfully flash the Firmware by making the Changes in user_config.h and enabled HAVEENC28J60. I have a project where i needed a simple Ethernet to WiFi Router to allow my other ESP's to be able to connect to it. 
I have made the Following changes:
set eth_enable 1
set eth_dhcpd 1
set speed 160

When I type 
show:
I get to see the following:
Version V2.2.15 (build: Sun Mar 14 13:22:44 2021)
STA: SSID:ssid PW:password [AutoConnect:0]
AP:  SSID:MyAP  PW:none [open] IP:192.168.4.1/24 [NAT]
ETH IP: 192.168.5.1 Netmask: 255.255.255.0 Gateway: 0.0.0.0
STA MAC: 5c:cf:7f:53:c6:a5
AP MAC:  5e:cf:7f:53:c6:a5
ETH MAC: 5a:cf:7f:53:c6:a5
STA hostname: ESP_53c6a5
Network console access on port 7777 (mode 3)
Clock speed: 160

Show route:
Routing table:
Network              Dest
192.168.4.0/24       ew1
192.168.5.0/24       en0
127.0.0.0/8          lo0
default              192.168.4.1

Serial Monitor shows the Following:
Boot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz

Booting rom 0.
⸮⸮⸮⸮o⸮r⸮⸮g|⸮dldl`b⸮⸮|{⸮l⸮n⸮⸮o⸮
WiFi Repeater V2.2.15 starting

running rom 0
Config found and loaded (2836 Bytes)
Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
Starting enc28j60
Starting enc28j60 DHCPd
mode : sta(5c:cf:7f:53:c6:a5) + softAP(5e:cf:7f:53:c6:a5)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
add 1
aid 1
station: 40:a3:cc:bd:aa:1a join, AID = 1

The Problem is that I am unable to access the Internet. When I visit 1192.168.5.1 i get to see the NAT router page. But no internet connectivity. Can someone guide me as to where i am going wrong?

I've made the connections as per the Instructions. Previously I had connected the ENC28j60 CLK to D8. Turns out that was causing the ESP to bootloop. When i switched it to SCK it started working. I am unsure where I've gone wrong.

Regards,
Saket

martin-ger, 2021-03-15:
What does "show status" print?
What happen if you give "interface if0 down"?

slowpoison4, 2021-03-16:
Hi Martin,
This is what i get when i type show stats. show status returns "Invalid command".
show stats
System uptime: 0:01:43
8 KiB in (73 packets)
0 KiB out (4 packets)
Power supply: 2.911 V
Phy mode: n
Free mem: 7512
STA not connected
ETH IP: 192.168.5.1/24 GW: 0.0.0.0
0 Stations connected to SoftAP

When I Send the command "interface if0 down" it returns with Invalid interface

Regards,

martin-ger, 2021-03-16:
Could you try to set a default gw address for the ethernet (the upstream internet router) with:

set eth_gw x.x.x.x

slowpoison4, 2021-03-16:
Hi Martin,

I set the default gateway to 192.168.0.1. 

when I input show:
Version V2.2.15 (build: Sun Mar 14 13:22:44 2021)
STA: SSID:ssid PW:password [AutoConnect:0]
AP:  SSID:MyAP  PW:none [open] IP:192.168.4.1/24 [NAT]
ETH IP: 192.168.5.1 Netmask: 255.255.255.0 Gateway: 192.168.0.1
STA MAC: 5c:cf:7f:53:c6:a5
AP MAC:  5e:cf:7f:53:c6:a5
ETH MAC: 5a:cf:7f:53:c6:a5
STA hostname: ESP_53c6a5
Network console access on port 7777 (mode 3)
Clock speed: 160

show stats:
System uptime: 0:01:09
2 KiB in (22 packets)
1 KiB out (4 packets)
Power supply: 2.914 V
Phy mode: n
Free mem: 7232
STA not connected
ETH IP: 192.168.5.1/24 GW: 192.168.0.1
1 Station connected to SoftAP
Station: d8:a3:15:44:17:ed - 192.168.4.2

show route
Routing table:
Network              Dest
192.168.4.0/24       ew1
192.168.5.0/24       en0
127.0.0.0/8          lo0
default              192.168.4.1

Still no access to internet. If you see, the Ethernet IP shows 192.168.5.1. When i enter this address in my mobile phone browser, it takes me to the NAT Settings page.

martin-ger, 2021-03-16:
The ethernet interface must have an IP in the subnet of the connected upstream router (192.168.0.x) and 192.168.0.1 as default gw. Than it should be able to ping to the outside world. NAT should than enable Internet access for the connected STAs.

slowpoison4, 2021-03-16:
Hi Martin,

I checked the DHCP client list on my router and couldn't find the ethernet module. So i don't think the Module has initialized correctly. What can i do to ensure it is initialized correctly? do you by any chance have a working version with the stats that i could compare it to. also what setting can i further change to just get it running ? 

Regards

martin-ger, 2021-03-16:
There is no DHCP client running on the ethernet interface. Just give it a static IP (best out of the range of the DHCP server in the uplink router), but for a try any unused IP will do). Currently I have no config running here.

ulsanjeewa, 2021-06-21:
> Hi
> I have a NODEMCU ESP8266 and an ENC28J60 Ethernet Module.
> 
> I've been able to successfully flash the Firmware by making the Changes in user_config.h and enabled HAVEENC28J60. I have a project where i needed a simple Ethernet to WiFi Router to allow my other ESP's to be able to connect to it.
> I have made the Following changes:
> set eth_enable 1
> set eth_dhcpd 1
> set speed 160
> 
> When I type
> show:
> I get to see the following:
> Version V2.2.15 (build: Sun Mar 14 13:22:44 2021)
> STA: SSID:ssid PW:password [AutoConnect:0]
> AP: SSID:MyAP PW:none [open] IP:192.168.4.1/24 [NAT]
> ETH IP: 192.168.5.1 Netmask: 255.255.255.0 Gateway: 0.0.0.0
> STA MAC: 5c:cf:7f:53:c6:a5
> AP MAC: 5e:cf:7f:53:c6:a5
> ETH MAC: 5a:cf:7f:53:c6:a5
> STA hostname: ESP_53c6a5
> Network console access on port 7777 (mode 3)
> Clock speed: 160
> 
> Show route:
> Routing table:
> Network Dest
> 192.168.4.0/24 ew1
> 192.168.5.0/24 en0
> 127.0.0.0/8 lo0
> default 192.168.4.1
> 
> Serial Monitor shows the Following:
> Boot v1.4.2 - [richardaburton@gmail.com](mailto:richardaburton@gmail.com)
> Flash Size: 32 Mbit
> Flash Mode: DIO
> Flash Speed: 40 MHz
> 
> Booting rom 0.
> ⸮⸮�⸮�⸮o⸮r⸮⸮g|⸮d�ldl`��b⸮���⸮|�{⸮�l⸮n⸮�⸮o⸮
> WiFi Repeater V2.2.15 starting
> 
> running rom 0
> Config found and loaded (2836 Bytes)
> Starting Console TCP Server on port 7777
> Starting Web Config Server on port 80
> Starting enc28j60
> Starting enc28j60 DHCPd
> mode : sta(5c:cf:7f:53:c6:a5) + softAP(5e:cf:7f:53:c6:a5)
> add if0
> add if1
> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
> bcn 100
> add 1
> aid 1
> station: 40:a3:cc:bd:aa:1a join, AID = 1
> 
> The Problem is that I am unable to access the Internet. When I visit 1192.168.5.1 i get to see the NAT router page. But no internet connectivity. Can someone guide me as to where i am going wrong?
> 
> I've made the connections as per the Instructions. Previously I had connected the ENC28j60 CLK to D5. Turns out that was causing the ESP to bootloop. When i switched it to SCK it started working. I am unsure where I've gone wrong.
> 
> Regards,

Can you help me?
I have only enabled the ENC28J60 in the user_config.h (#define HAVE_ENC28J60 1)and get the following error: 

Linux terminal :
user@user-virtual-machine:/Desktop/project2/esp_wifi_repeater-master$ make clean
user@user-virtual-machine:/Desktop/project2/esp_wifi_repeater-master$ make
CC driver/spi.c
CC driver/new_uart.c
CC user/config_flash.c
CC user/ringbuf.c
CC user/rboot-ota.c
CC user/acl.c
CC user/user_main.c
CC user/rboot-api.c
CC user/sys_time.c
CC mqtt/mqtt.c
CC mqtt/mqtt_msg.c
CC mqtt/queue.c
CC mqtt/utils.c
CC mqtt/ringbuf_mqtt.c
CC mqtt/proto.c
CC easygpio/easygpio.c
AR build/app_app.a
/home/sanjeewa/Desktop/project2/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: build/app.out section .text' will not fit in region iram1_0_seg'
collect2: error: ld returned 1 exit status
Makefile:132: recipe for target 'firmware/0x02000.bin' failed
make: *** [firmware/0x02000.bin] Error 1

Thanks
Sanjeewa.

milad-bagheri, 2023-02-12:
hi how did you enabled HAVEENC28J60 i forked the project and changed it and built firmware but it dose not work at all!!!!