martin-ger/esp_wifi_repeater #444: Can I add an oled display?

classic-audio, 2021-03-27:
I have flashed the software to Wemos D1 Mini Pro and all works great.
Can I now add an Arduino Ide sketch also?
Because:
I would like to use the extender as an improved wifi network to Tello (drone).
I will use a powerbank as psu.
I would like to know the battery status by using the oled display (if possible)

martin-ger, 2021-03-27:
It is not trivial to add an oled display to an esp_open_sdk project. 

It is probably easier, if you start from this example in Arduiono: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/RangeExtender-NAPT/RangeExtender-NAPT.ino

classic-audio, 2021-03-27:
Thank you very much for a very fast reply.
I expected this reply.
And you even gave me a suggestion - thanks - I will follow that.

closed 2021-03-27 by classic-audio