martin-ger/esp_wifi_repeater #445: ENC28J60 Support ,iram1_0_seg Error

GrafvonWolfenstein, 2021-03-27:
Hello Martin,

I have only enabled the ENC28J60 in the user_config.h and get the following error:

`.text' will not fit in region `iram1_0_seg'

Do you know this error? Can you help me?
Do you have a compiled image with the ENC28J60 enabled?

Greetings
GrafvonWolfenstein

ulsanjeewa, 2021-06-21:
Same thing happened to me.

ENC28J60 enabled  in the user_config.h  (#define HAVE_ENC28J60 1)

Linux terminal :
user@user-virtual-machine:~/Desktop/project2/esp_wifi_repeater-master$ make clean
user@user-virtual-machine:~/Desktop/project2/esp_wifi_repeater-master$ make
CC driver/spi.c
CC driver/new_uart.c
CC user/config_flash.c
CC user/ringbuf.c
CC user/rboot-ota.c
CC user/acl.c
CC user/user_main.c
CC user/rboot-api.c
CC user/sys_time.c
CC mqtt/mqtt.c
CC mqtt/mqtt_msg.c
CC mqtt/queue.c
CC mqtt/utils.c
CC mqtt/ringbuf_mqtt.c
CC mqtt/proto.c
CC easygpio/easygpio.c
AR build/app_app.a
/home/sanjeewa/Desktop/project2/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: build/app.out section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
Makefile:132: recipe for target 'firmware/0x02000.bin' failed
make: *** [firmware/0x02000.bin] Error 1

Thanks
Sanjeewa.

GiVeMeRoOt, 2021-08-12:
Hey @GrafvonWolfenstein and @ulsanjeewa I was not able to fix this, however I found out an alternative solution:
1. Fork this repo: https://github.com/shafr/esp_wifi_repeater
2. Make your changes to the code and push to your repo
3. You should see builds happening in "Actions" tab.

If you just want the ENC option enabled, I have already compiled it in my repo: https://github.com/GiVeMeRoOt/esp_wifi_repeater/actions/runs/1117319242 (firmware.tar.gz on this page)