martin-ger/esp_wifi_repeater #446: How to use WEP for Soft AP?

programings, 2021-03-29:
How can i switch the security standard / scheme to WEP for the Soft AP?

I would like to use the router as a teaching tool to present some WEP security vulnerabilities.

I guess it is a trivial change in the code, however i'm not familiar at all with the ESP SDK.

KHUSAINIBINAJ, 2021-03-30:
I also don't understand about technology. but i love it. I am self-taught.

I am currently having trouble using the wireless internet access using my new comfast cf 130n tool.

Tried using the Telkomsel PPPoe user and password, the WiFi ID has not been connected yet.

do you have isp free for me

stefanbode, 2021-03-30:
WEP is as far as I know never implemented. It was already insecure when the ESP came out. Take a raspberry pi for this

KHUSAINIBINAJ, 2021-04-04:
I have an orange pi device so far I have not operated it. because it is still busy connecting to the internet for free with comfast cpe

KHUSAINIBINAJ, 2021-04-04:
besides that, the keyboard mouse, hdmi cable and micro sd, I haven't bought it yet