martin-ger/esp_wifi_repeater #448: Avoid Regular clients in Automesh on ESP APs due to performance issues

Basti-Fantasti, 2021-03-31:
Hi,

when setup in Automesh mode the ESPs AP has the same SSID as the uplink Router.
Is there a way to prevent devices from connecting to the ESPs AP due to the lower performance?

I want to use it only for sensors not phone or notebooks

Thanks in advance

Basti-Fantasti, 2021-04-01:
Thanks for your quick reply.

Yes I think I can handle the setup of the sensors to have a setup like this (if this is what you suggested):

- Router IP Range 192.168.0.1/24, SSID: **MyHomeNetwork**
- 1st ESP as "regular nat router" configured with Station SSID: MyHomeNetwork (for uplink) and self published SSID: **MyIoTNet** quite close to the router
- ESP 2-n configured in **Automesh mode** with the 1st ESPs published SSID **MyIoTNet** as Station

This could be a good solution to have a separate (lower performance) mesh network for sensors and actors around the house.

A point is, that I cannot tell, to which AP the sensors connect to, which forces me to use DHCP on the sensors to acquire an IP address due to the different IP subnets. I thought about settings my battery driven sensors to static IPs to reduce the power usage and up time between deep-sleeps in the hand shake and address assigning process.

In this scenario I would only need to have a static route for all potential IP ranges 10.24.0.0/16 to the static IP of the 1st ESP

Does this make sense so far 😄 

Best regards
Bastian

martin-ger, 2021-04-01:
I still don't get, where you need the static routes. All clients can reach any upstream node. If eg. an MQTT server ist located in the "main" net  at192.168.0.x it can be accessed by all nodes in the Automesh.

Basti-Fantasti, 2021-04-01:
but is it also possible to use actors (switches) in the meshed network? So not only sending data to my HA Server in the 192.168.0.x is possible but also from server to the actors without the route?

martin-ger, 2021-04-01:
No, that is not - relaying via a central MQTT might be a solution...

Basti-Fantasti, 2021-04-01:
Ok, so I need to check if this is possible. I have ESPHome or Tasmota running on my switches. For Tasmota this should be ok. They are connected to my Mosquitto on my HA installation - but I need to check for the ESPHome devices.