martin-ger/esp_wifi_repeater #449: Can't find the "D1 mini pro" in my Fritzbox under active network connections.

TomComm27, 2021-04-02:
In the "D1 mini pro" under the "STA Settings" I have the
SSID: Fritz
Password: xxxxxxxxxxxxxx
entered and clicked Connect.

Now I can still see the access point under my WLANs and also get to the settings under 192.168.4.1.
In my "Fritzbox 6590 Kabel" the "D1 mini pro" does not appear as an access point. This should already be the case, right?

If I now activate the automesh and restart the D1 mini, I can no longer find it under my WLAN´s and it still does not appear in the Fritzbox under active network connections.
I guess as long as the Fritzbox has not added the D1 mini pro as an AP or WiFi device, it will not work !?

I have a mash active with a Fritzbox repeater on the Fritzbox.
I set the encryption from "WPA2 + WPA3" to "WPA2".

I hope you can help me. It is really a brilliant project.

martin-ger, 2021-04-02:
It should appear as a connected Station. Is the ESP blinking? (good if so, then it thinks, it is connected).  

If not, any issues with the PW? Try first to connect without "Automesh"

TomComm27, 2021-04-02:
No, it doesn't blink when I supply it with energy.

Thanks to you I have the solution:
The password was correct, but the SSID is case-sensitive.
The "FRITZ" must be completely capitalized so that it works.

Now I see the "D1 mini" in my Fritzbox.

Now I will try the Automesh function.