martin-ger/esp_wifi_repeater #452: Problem with connecting to FRITZ!Box

Mueller1842, 2021-04-18:
Hi,

After flashing, I could see the MyAP, can connect to it and I can use the adress 192.178.4.1.
My router is an fritzbox with SSID "FRITZ!Box 7580 RB" (original name).

First of all, I typed the SSID and key (and verified it), but after restarting the ESP, it's LED is on all the time.
In the Fritzbox, the ESP isn't connected.

How can I solve the problem?

martin-ger, 2021-04-18:
I am not aware of any specific problems with Fritzboxes - my guess would still be an Issue with the name. Could you try entering and verifing it via the cli (telent 7777 or serial)?

Mueller1842, 2021-04-18:
Sorry, my knowledge isn't so good in these things.

I would like to try it with %20 in stead of " " in the name. But even after flashing with other programms en reinstalling die AP-files: the MyAP doesn't appear any more. This problem I had bevor too. Wy it appears this morning - and now it doesn't appear - I don't know.