martin-ger/esp_wifi_repeater #454: Wifi name with _ not work on webpage

dmolner, 2021-04-26:
Hello,

I test the firmware and when i put xxx_xxx in SSID not work
I put in Arduino IDE Monitor serial the same and work correctly.

I can't work it in D1 mini

martin-ger, 2021-04-26:
WPA2 password must be at least 8 chars.

dmolner, 2021-04-26:
My SSID is xxxx_xxxxx 10 caracters

elijash, 2022-02-18:
> WPA2 password must be at least 8 chars.
>
>
>
> Von: dmolner ***@***.***
> Gesendet: Montag, 26. April 2021 10:01
> An: martin-ger/esp_wifi_repeater
> Cc: Subscribed
> Betreff: [martin-ger/esp_wifi_repeater] [...]

same problem with AP using space character