martin-ger/esp_wifi_repeater #462: Monitor service crashes ESP8266 every time

LZ2DMV, 2021-06-15:
Hi,

I'm trying to use the monitor service, but it crashes the ESP8266 every time. I guess the load is too much, although there is only one client connected. Is there a specific way this feature is meant to work?
My main interest is to monitor the traffic for a single client, local to the ESP LAN, that goes to a specific port (destination IP could be local or external).

Just to see if its working, i first tried to start the monitor service to monitor all the traffic, and that crashed the ESP the moment i connected to the listening socket.
Then, i tried using ACLs to limit the amount of data that should be processed and sent over the air, so i did:

`acl from_sta TCP 192.168.4.2 any any 80 allow_monitor`

Again, that crahes the ESP. Then, i tried limiting the scope even more:

`acl from_sta TCP 192.168.4.2 any 192.168.4.1 80 allow_monitor`

But, unfortunately, the same result.

martin-ger, 2021-06-15:
Can confirm, that there is a problem right now. I will look into this. If you just want to have the functionality, you might try to use an older version, e.g.` git checkout c2e0d9947225799a836e5feb0bc7fac4bbd48021` works for me.

LZ2DMV, 2021-06-16:
It still doesn't work for me with that revision either, and even with older ones. :( What i have noticed is that Monitor mode works fine as long as we are not connected to an uplink router. Immediately after that happens (with Monitor mode on), the ESP crashes.

Also, starting Monitor mode with ACLs doesn't work no matter if we have an uplink or not, i.e.:

`acl from_sta TCP 192.168.4.2 any 192.168.4.1 80 allow_monitor`
`monitor acl 1337`

After that, the ESP crashes.