martin-ger/esp_wifi_repeater #467: MQTT - Sleep - clean session, subscribing to retained sleep message when waking

Cloolalang, 2021-06-30:
I thought I might be able to send a retained message to the command topic: sleep 3600, and the Nat would sleep, wake hourly and then subscribe to the retained message and sleep again.
I cant get that to work, I tried setting QOS to 1 as well.
Im wondering if the NAT MQTT client is connecting in clean session mode? and therefor not subscribing to the retained messages?

Is there a way to change this, did I miss something?
So now I want the Nat Repeaters to sleep all night, so I have to keen sending sleep 3600 very hour though the night?
P

martin-ger, 2021-06-30:
The client connect with clean session. You can change that in the sources in user/user_main.c#L4511

Cloolalang, 2021-07-03:
> 
> 
> The client connect with clean session. You can change that in the sources in user/user_main.c#L4511

Thank-you Martin for the help with this information. 
Since I am not setup for making builds, I made a workaround automation in Home Assistant, whereby at night HA sensor listens to the LWT topic and when it goes online, HA publishes sleep 3900 to the command topic.
3900 ( not 2600) is like an hour on my nodeMCU for some reason.
Worth noting that the MQTT timeout is 3 mins, so it take 3 minutes to show as offline in HA.

All good, its working very well!
P