martin-ger/esp_wifi_repeater #469: NTP sync of repeater sleep

steenl, 2021-07-28:
Hi, this may just be a documentation request, but am not clear from other issue resolutions.
Ideally the repeater would NTP sync prior to going to sleep such that my sensor can wake every hour and be assured a repeator access to post data (and endpoint do NTP sync)

Is the NTP sync server setting static, and how to enable would be helpful documentation

Thanks, and excellent project

martin-ger, 2021-07-28:
There is currently no NTP sync for sleep timing. It would only help to start at a defined time, but the biggest source of unpredictability for the wakeup is the drift of the internal clock, which cannot be influenced by the NTP.

The idea for a synchronized sleep is currently an MQTT-message coming from a synchronizing source,

steenl, 2021-07-28:
Thanks for response.  

It would seem one way to avoid requiring a synchronizing source is to allow NTP sync on all devices (repeators and sensors) and accommodate for a small drift.  For example, say I have 10 beehives monitoring weight/temp/humidity hourly with ESP8266 that need one repeater to access the internet.  If each device could have up to a +/-1second drift per hour, and the repeater wakes up >2 seconds earlier (with some margin on DHCP and other initialization delays), then (assuming all 11 devices NTP sync successfully) it seems a stable solution could be made.  The repeater would need to be configured to be awake for both negative drift as well as positive drift account for both repeater and sensor sources.  
Agreed that some characterization would be needed to quantify the drift on both repeater and sensor, but seems like a useful feature.  I can help code/debug/validate repeater code changes, but my background is more in hardware.