martin-ger/esp_wifi_repeater #470: Portmap failed

hermannw, 2021-08-09:
I installed the software successfully on an ESP8266 and configured it.
From what I understand, I have to do some port mapping in order to be able to see another ESP8266 that is behind the repeater. Similar to [this ](https://github.com/martin-ger/esp_wifi_repeater/issues/216)topic. 
Here are a few things I can read via telnet:
CMD>show stats
System uptime: 21:14:46
5869 KiB in (24194 packets)
3378 KiB out (23939 packets)
Power supply: 3.389 V
Phy mode: n
Free mem: 27272
STA IP: 192.168.1.216/24 GW: 192.168.1.1
STA RSSI: -71
1 Station connected to SoftAP
Station: 50:02:91:e9:f8:1f - 10.24.1.3

CMD>show route
Routing table:
Network              Dest
10.24.1.0/24         ew1
192.168.1.0/24       ew0
127.0.0.0/8          lo0
default              192.168.1.1

set nat is 1.
So in order to see the ESP that is connected to 10.24.1.3 I assume I have to do:
CMD>portmap add TCP 80 10.24.1.3 80

Result: Portmap failed.

CMD>portmap add TCP 80 10.24.1.3 80
Portmap failed

Did I miss something here?

hermannw, 2021-08-10:
Looking a bit further into this, I noticed that "show" and "show config" are showing the same information.
CMD>show
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535
Portmap: ???: 192.168.1.216:65535 -> 255.255.255.255:65535

Looking at other issues here, I see that "show" should look totally different.

I am using telnet. Is there any difference to the serial console?

hermannw, 2021-08-11:
Any ideas? I did a lot of searching and it seem this project is quite popular but nobody seems to have these problems. I tried to flash the software onto two other Wemos D1 mini and it's the same. The devices from behind the repeater can send the information to the Internet but I cannot access any of them directly. Portmapping always gives me the same error. Unfortunately I don't have the knowledge to compile it myself.

martin-ger, 2021-08-11:
currently on holyday - will look into it end august. not aware of an issue here,

hermannw, 2021-08-11:
No problem Martin. Have a good vacation!

hermannw, 2021-08-15:
Soooo, I got it working...
After numerous times deleting and flashing again it works now. Not really sure what the problem was.
I found an old ESP-01 and decided to try that. I used these settings:

[image]

It worked immediately. Then I used the same settings - except the flash size (16Mbit) - for an ESP8266-12 (ESP8266MOD). It did not work. I had to erase and re-flash the module several times. My phone could connect to MyAP but the website never loaded. After several more erase cycles, I opened the serial console. It showed the status, waited a few seconds and then it automatically connected to the WiFi. Automesh was enable like before, the mqtt settings were still there, so they must have survives the erase... 
Show and show config show the correct information and port mapping works. 
It would still be interesting to know what the problem was...

Thanks again Martin! Hope you have a good vacation.
Hermann