martin-ger/esp_wifi_repeater #478: ESP32 Support

HARSHAL93, 2021-11-22:
Hi Martin,

Appreciated your work. As I am trying to create wi-fi repeater with ESP32 with the similar functionality. As I am having understanding of networking,ESP32, and ESP32 Mesh network. Can you please help me to where to start & which part we need to create from scratch in ESP32 if we are taking current lib as reference.
Basic requirement are below:
ESP32 Wi-Fi repeater with support of internet.
Mesh Network support 
basic web page support for configuration same as you provided in existing lib

martin-ger, 2021-11-22:
Starting point should be the https://github.com/martin-ger/esp32_nat_router . It contains all the required features aside from mesh. The mesh state mashine should be added - best from scratch. A detailed description can be fould in short in the README of this repo and longer here: https://ieeexplore.ieee.org/document/8795311 .

aircable, 2021-11-22:
I'm working on a Wifi mesh implementation, I call WISH. It is based on the
802.11 frames in the ESP32 framework and modeled after the old CSR mesh.
Since ESPnow is not available as source, I decided to implement it from
scratch.
I haven't published anything yet on Github. My framework may not be the
right environment for publication because it is the firmware for our
AIRcable products.

But I would like to get this out and find a better environment for that.
@martin-get Any idea how we could add mesh to your nat_router code? Is that
the right way?

The main problem I have is that I usually don't run AP/STA code and the
WISH code at the same time. The issue is that the WISH code uses multiple
channels to transmit and receive frames for reliability mainly. When
connected to an AP, the channel number is fixed and I can't change it
without disconnecting.

BTW, WISH uses a flooded mesh approach instead of an autotree routing
mechanism. No NAT necessary, but it requires promiscuous mode.

Interested?