martin-ger/esp_wifi_repeater #489: ESP Wi-Fi Repeater

kabhaya1, 2022-02-27:
Thanks martin-ger for the excellent project “ESP Wi-Fi Repeater”. I am a faculty in Dept. of Comp.Sc. & Engg. at Trident Academy of Technology, Bhubaneswar, Odisha, India.

We are using this for our IoT Projects. We want to extend this project with the Ethernet NIC Support. As per the link you have given for the “ESP WiFi Repeater” (https://github.com/martin-ger/esp_wifi_repeater), in order to activate the ENC28J60 Ethernet Support, the README suggests to Switch ON the HAVE_ENC28J60 compile option in "user_config.h" and re-build the code from scratch. I do not know how to build this and the source is also not given. In addition, we want to do some customization as below:
1. Change the Status LED Port
2. Include a Battery Status Monitoring Module that flashes LED before going to deep sleep
3. Customize the default configuration setting

We shall be extremely grateful if you can kindly guide us how to do this. As there is no source folder, we are unable to proceed further. Therefore, we are eagerly looking forward for your support.

Thanks for the awesome project.

With best regards,

Abhaya Kumar
My Email ID: abhaya@tat.ac.in, kabhaya1@gmail.com

NightFalcon650, 2022-06-05:
i'm having the same problem. Trying to make it with visual studio code but there's no platformio.ini file to make it with platformio and also non src directory. Could you expand the explanation with examples howto build it with visualstudio code?

G2G2G2G, 2022-06-14:
I just found this software about 15 minutes ago and ctrl+f build and found https://github.com/martin-ger/esp_wifi_repeater#building-and-flashing and built it in like 3 minutes. 
"the source is not given" what?