martin-ger/esp_wifi_repeater #490: No AP with ssid SSID found

JaimeGonz, 2022-03-02:
Hello, I am using the repeaters in Automesh mode. But I would like to know if it is possible that in case a repeater loses connection with the AP, it can be returned to AP mode (MyAP), to reconfigure it with new credentials (SSID and PASSWORD). 

Since when that happens, it goes into an endless loop trying to find the AP it was connected to. 

Thanks for your help.

[image]

martin-ger, 2022-03-02:
You can always reset it using the serial console. From there (or in user_config.h) you can also set this: https://github.com/martin-ger/esp_wifi_repeater#hw-factory-reset
However: default is 255, meaning "disbabled". But you can enable this via the serial terminal as well.

JaimeGonz, 2022-03-02:
Thank you Martin.

zerocold2, 2024-07-07:
I am facing the same issue , serial port can't respond to any command I am trying to reset factory but nothing affect

[image]