martin-ger/esp_wifi_repeater #496: Using shelly 1

AquaMCU, 2022-06-18:
Hello. 

I wonder, if the router would run on a shelly 1. This is a 8266 unit that contains a relay and switch and is pretty hackable. Since you can connect it directly to mains and it is about 20 USD, it is a perfect unit for my use cases. 

Here you can find the details:
https://shelly.cloud/knowledge-base/devices/shelly-1/

Oliver

martin-ger, 2022-06-18:
As long as it is an ESP8266, I'm pretty sure, it will run - the question is, what it should do with the relay?

RibeiroSantos, 2022-06-18:
Maybe allow to control the relay by mqtt or web commands. Lets say it would work as a repeater together with the relay Control.

AquaMCU, 2022-06-18:
Exactly. It is easy to install and has a relay and power supply.

BTW: It works! awesome.

RibeiroSantos, 2022-06-18:
> Exactly. It is easy to install and has a relay and power supply.
> 
> BTW: It works! awesome.

Now, check the uMQTTBroker[uMQTTBroker](https://github.com/martin-ger/uMQTTBroker)  repository, join all together and you have the perfect world!