martin-ger/esp_wifi_repeater #506: What was the reason forwhich  esp_wifi_repeater was created?

spaceman5, 2022-11-16:
Hi Martin

esp_wifi_repeater is an impressive project,
you literally implemented a Wireless Router, with some additional functionality that cannot be found on most routers (e.g. Port Mirroring), on the ESP8266.

I am curious to ask you a non-technical question please:
What was the reason for which you created it?

Thank you

martin-ger, 2022-11-16:
As I am doing R&D in the IoT/wireless networking area, I was curious about the ESP8266, as it was so cheap, powerful and reasonably well documented. I was simply playing around a little bit and most things worked exceptionally well. From there it evolved... ;-)

spaceman5, 2022-11-16:
Quite a deep evolution.. ;)

BTW, If I can recommend you,
If you need Port Mirroring for Ethernet,
a really nice Smart Switch is the TP-Link TL-SG108E.
https://www.amazon.com/dp/B00K4DS5KU/

It only costs 30$.

Speaking of Ethernet, you mentioned in the ReadMe support for the ENC28J60 Ethernet Module.

I remember back in the days when I started with Arduino (it was 2012),
there were 2 Ethernet Modules available back then:
Wiznet W5100,
and the ENC28J60.

The Wiznet one, back then, cost 20$,
and the ENC28J60 was about 3$.
But in every Forum that I read about the ENC28J60, users wrote
"don't buy it, it's problematic, only buy the Wiznet one".
It wasn't just one thread, but many like that..

So when you worked with the ENC28J60, was it working OK for you?

BTW, since then, the price of the Wiznet W5100 dropped, and is now just a few dollars.

martin-ger, 2022-11-16:
The Wiznet W5100 is somewhat similar to the original SW of the ESP8266. It is not only a NIC but a complete controller with Ethernet MAC and PHY plus a basic TCP/IP stack. Obviously this make it much easier for an Arduino ATMega to handle simple TCP-communication via Ethernet. However, if you have an ESP with a full LWIP-Stack (and already existing IP-connectivity), it only makes sense to and another MAC and PHY - like the ESP32 already includes an Ethernet MAC, that only needs an additional PHY. 

The ENC28J60 support is still beta - some simple tests work, but it is not stable under load (buffer issues?). I stopped working on that, as there was no real use case and interest in the community. Today, if you really need WiFi and Ethernet on a µC, you simply use an ESP32, it is more performant and even cheaper.

Thanks for the hint with the TP-Link.

spaceman5, 2022-11-16:
Oh right, I now recall all about the differences between the W5100 and the ENC28J60.

Another recommendation If I may,
TP-Link has a really nice and small router, which is called the TL-WR844N.
https://www.tp-link.com/en/home-networking/wifi-router/tl-wr844n/v1/

It is quite a weird animal, compared to most of TP-Link's other routers:
It's a compact router, yet it's not a portable router, since it has quite big Wi-Fi antennas.
It works on 5v 0.6A (one of their few Routers that work on 5v, the other one I know is their Portable Router model, which lately is not seen in stores).
It is Multi-Mode, like all of their routers, but in addition to the modes Router/AP/Repeater, it includes a WISP mode,
which is basically what you do with the ESP here - connect to the Internet (uplink) via Wi-Fi.

And, like all of TP-Link's Home Routers, it does not have Port Mirroring sadly.
So your great project will remain useful for many people for quite a long time.

closed 2022-11-17 by spaceman5