martin-ger/esp_wifi_repeater #509: Error while connecting to WPA Enterprise network

kbshal, 2023-01-14:
It gives an error while trying to connect to the enterprise network

vakshit, 2023-01-19:
how do you check for errors?
i'm having the same issue. I'm able to connect to a WPA2 network but not able to connect to enterprise network.

martin-ger, 2023-01-19:
ESP8266 and WPA Enterprise  is a pain. Think, often the certificates used by the WPA Enterprise networks are too large for the ESP8266's memory and this causes crashes. If possible, try with the ESP32 instead.

vakshit, 2023-01-19:
my enterprise network doesn't use CA certificates. if that's what you are talking about.

martin-ger, 2023-01-19:
What kind of authentication do you use?

vakshit, 2023-01-19:
[image]

this config after entering my username and password works well on my system.

martin-ger, 2023-01-19:
This only means, this config doesn't check the server's certificate with a CA cert - there is the chance that you connect to a fake AP. Still the RADIUS server has a cert and sends it. See also: https://security.stackexchange.com/questions/182530/is-it-possible-for-an-802-1x-network-peap-mschapv2-to-have-no-certificate

martin-ger, 2023-01-19:
MSCHAPv2 itself isn't a problem here as it is used inside a TLS tunnel to check the PW.

vakshit, 2023-01-19:
Yes, there is a chance that the RADUIS server sends me a CA certificate, but i dont know a way to check it. Also, it is my institute's wifi so it's not getting connected to a fake AP. 

Shall I try on an ESP32? I'm actually still unable to find out the issue. 😅

martin-ger, 2023-01-19:
Think, you should try.

vakshit, 2023-01-19:
the same code or this one [https://github.com/martin-ger/esp32_nat_router](https://github.com/martin-ger/esp32_nat_router).

abinthomas744, 2023-03-10:
> how do you check for errors? i'm having the same issue. I'm able to connect to a WPA2 network but not able to connect to enterprise network.

same question, I'm looking for a regularly flashing light to indicate an uplink but can't see it other than that no errors as such