martin-ger/esp_wifi_repeater #516: Updated UI - Code

jcjms, 2023-05-13:
Dear Martin-ger, 
I am a big fan of this project and have been using it for years. That's why I decided to build a simple UI. Because of problems with Docker, I can't test whether it works properly with the esp and therefore don't want to create a pull request. I hope you like it. If you want to use it would appreciate some form of credit. Kind regards, jcjms.

<div style="display: flex;">
[image]
[image]
</div>

PS: the heart in the footer bugs a bit because of my server. The HTML file itself works great. Also the MAC address is a dummy address.

[web.h.txt](https://github.com/martin-ger/esp_wifi_repeater/files/11469853/web.h.txt)
[styles.css.txt](https://github.com/martin-ger/esp_wifi_repeater/files/11469854/styles.css.txt)
[image]
[image]
[image]
[image]
[image]

VilotStar, 2023-05-30:
You would need to have the files hosted by the board as well or just hardcode them directly into the web.h right?

jcjms, 2023-06-04:
@VilotStar hardcoding them into the web.h should work