martin-ger/esp_wifi_repeater #527: After Flash Wemos mini D1 => no Accespoint found

hasenradball, 2024-01-07:
Hi,

I tired you app as repeater with a Wemos mini D1.
Flshing works fine, but after flash is finished and reset of the chip no SSID "MyApp" is found.

Is this a known issue?

l3ifk, 2024-01-28:
I have the exact same issue with an ESP-01. Flashing shows no errors but there it no Access Point called MyApp...

hasenradball, 2024-01-29:
@l3ifk no Found no solution.

I used the sketch „ range extender“ from the Arduino IDE examples.

l3ifk, 2024-01-29:
Thank you for your quick reply. Where exactly can I find the example code. I looked at all the ESP examples and did not find a range extender example...

hasenradball, 2024-02-01:
Sorry for the late reply.

Normally under the examples under wifi for the esp8266 wifi.

simontpellier, 2024-07-26:
> @l3ifk no Found no solution.
> 
> I used the sketch „ range extender“ from the Arduino IDE examples.

Thanks for the trick !
I should add that I then had to solve a “limited connection” issue. With an easy soluce here  (windows) : https://www.youtube.com/watch?v=uDj7a5JH1E4
and the whole works like a charm

hasenradball, 2024-07-26:
@simontpellier 
what trick?
I did not get to work…

hasenradball, 2024-07-27:
@simontpellier the video is only in french did not understand whats going on there.

simontpellier, 2024-07-27:
...the trick? instead of Martin's binaries, to use the “RangeExtender-NAPT” sketch as given in the examples.

My config:
W10
Arduino 1.8.19
board "LOLIN(WEMOS) D1 mini Pro" ( [https://fr.aliexpress.com/item/1005005933528007.html?spm=a2g0o.order_list.order_list_main.106.2dd45e5bapEqTe&gatewayAdapt=glo2fra](url) )

The sketch:

_#if LWIP_FEATURES && !LWIP_IPV6

#define HAVE_NETDUMP 0

#ifndef STASSID
#define STASSID "mynetwork"
#define STAPSK "mynetworkpassword"
#endif

#include <ESP8266WiFi.h>
#include <lwip/napt.h>
#include <lwip/dns.h>

#define NAPT 1000
#define NAPT_PORT 10

#if HAVE_NETDUMP

#include <NetDump.h>

void dump(int netif_idx, const char* data, size_t len, int out, int success) {
  (void)success;
  Serial.print(out ? F("out ") : F(" in "));
  Serial.printf("%d ", netif_idx);

  // optional filter example: if (netDump_is_ARP(data))
  {
    netDump(Serial, data, len);
    // netDumpHex(Serial, data, len);
  }
}
#endif

void setup() {
  Serial.begin(115200);
  Serial.printf("\n\nNAPT Range extender\n");
  Serial.printf("Heap on start: %d\n", ESP.getFreeHeap());

#if HAVE_NETDUMP
  phy_capture = dump;
#endif

  // first, connect to STA so we can get a proper local DNS server
  WiFi.mode(WIFI_STA);
  WiFi.begin(STASSID, STAPSK);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    delay(500);
  }
  Serial.printf("\nSTA: %s (dns: %s / %s)\n", WiFi.localIP().toString().c_str(), WiFi.dnsIP(0).toString().c_str(), WiFi.dnsIP(1).toString().c_str());

  // By default, DNS option will point to the interface IP
  // Instead, point it to the real DNS server.
  // Notice that:
  // - DhcpServer class only supports IPv4
  // - Only a single IP can be set
  auto& server = WiFi.softAPDhcpServer();
  server.setDns(WiFi.dnsIP(0));

  WiFi.softAPConfig(  // enable AP, with android-compatible google domain
    IPAddress(172, 217, 28, 254), IPAddress(172, 217, 28, 254), IPAddress(255, 255, 255, 0));
  WiFi.softAP(STASSID "extender", STAPSK);
  Serial.printf("AP: %s\n", WiFi.softAPIP().toString().c_str());

  Serial.printf("Heap before: %d\n", ESP.getFreeHeap());
  err_t ret = ip_napt_init(NAPT, NAPT_PORT);
  Serial.printf("ip_napt_init(%d,%d): ret=%d (OK=%d)\n", NAPT, NAPT_PORT, (int)ret, (int)ERR_OK);
  if (ret == ERR_OK) {
    ret = ip_napt_enable_no(SOFTAP_IF, 1);
    Serial.printf("ip_napt_enable_no(SOFTAP_IF): ret=%d (OK=%d)\n", (int)ret, (int)ERR_OK);
    if (ret == ERR_OK) { Serial.printf("WiFi Network '%s' with same password is now NATed behind '%s'\n", STASSID "extender", STASSID); }
  }
  Serial.printf("Heap after napt init: %d\n", ESP.getFreeHeap());
  if (ret != ERR_OK) { Serial.printf("NAPT initialization failed\n"); }
}

#else

void setup() {
  Serial.begin(115200);
  Serial.printf("\n\nNAPT not supported in this configuration\n");
}

#endif

void loop() {}_

hasenradball, 2024-07-27:
got it
😊

l3ifk, 2024-07-27:
@simontpellier 
What speeds do you get with the extender? I just tried the code and it works great but I only use a ESP 8266 01 module and I get maximum speeds of 2,5mbit/s while I get >400mbit/s with my regular wifi. 
I know that it's just an ESP but I wondered if your board did better there

simontpellier, 2024-07-27:
here the results for me (DLS... not fiber !) regular vs extended (with external antenna)

[image]
[image]

simontpellier, 2024-07-27:
better (thank for the idea !), with an ESP Wroom 32 **WITHOUT** external antenna
(and other code (bin) : [https://github.com/dchristl/esp32_nat_router_extended](url)

[image]

tmsd2001, 2025-04-20:
Check Flash DIO Mode, my Wemos D1 with Flash C8h worked like this.