martin-ger/esp_wifi_repeater #537: Some DHCP question not answered in documentation.

Holgi0811, 2024-05-18:
Hi guys,
I have some dhcp related questions:
- Is it possible to reset dhcp leases?
- What is dhcp range, full subnet?
- Is it possible to use fixed dhcp addresses?
- Is there a bug in Version V2.2.16? "save dhcp" command is not working. Device is receiving a new ip every day on wake up.

Thank you

hmueller01, 2025-02-03:
Hi Holgi ;-)

_What is dhcp range, full subnet?_ -> No:
```
    dhcp_lease.start_ip = config.network_addr;
    ip4_addr4(&dhcp_lease.start_ip) = 2;
    dhcp_lease.end_ip = config.network_addr;
    ip4_addr4(&dhcp_lease.end_ip) = 128;
    wifi_softap_set_dhcps_lease(&dhcp_lease);
```

_Is it possible to use fixed dhcp addresses?_
README.md says only by saving the DHCP table.

If you save your DHCP table (`save dhcp`) ESP comes up with lease time set to 100000 after reboot.
If a device connects, the lease time gets set to 120 = 2h. So after a day it unfortunately gets a new IP.

I plan to open a PR to make the lease time changeable ...

hmueller01, 2025-02-03:
From ESP SDK documentation:
`bool wifi_softap_set_dhcps_lease_time(uint32 minute)`
Set ESP8266 SoftAP DHCP server lease time, 120 minutes by default.

hmueller01, 2025-02-05:
See #549