martin-ger/esp_wifi_repeater #543: Will someone help me?

liruibingniubi, 2024-10-26:
我叫李瑞兵,来自中国山西运城.
我想找个国际友人,手把手教我如何修改他的配网页面,并将本项目编译为bin文件,感激不禁,非常感谢.我的邮箱2893620314@qq.com

My name is Li Ruibing, from Yuncheng, Shanxi, China. I would like to find an international friend who can teach me how to modify his web page and compile this project into bin file. Thank you very much. My email is 2893620314@qq.com

Karthik7090ps, 2025-02-06:
> 我叫李瑞兵,来自中国山西运城. 我想找个国际友人,手把手教我如何修改他的配网页面,并将本项目编译为bin文件,感激不禁,非常感谢.我的邮箱[2893620314@qq.com](mailto:2893620314@qq.com)
> 
> My name is Li Ruibing, from Yuncheng, Shanxi, China. I would like to find an international friend who can teach me how to modify his web page and compile this project into bin file. Thank you very much. My email is [2893620314@qq.com](mailto:2893620314@qq.com)

Hi Li Ruibing, Its easy to compile if you understand the readme and the files in it to compile and create binaries, if you are still facing problems with understanding or compiling you can reach me out at karthikreddyr02@gmail.com

ical-lubis, 2025-11-24:
Hi. I'm using Windows 10, and my internet connection is very limited to install Docker, and my PC memory is also very limited.

Can someone help me compile this?
I want several features:
#define DAILY_LIMIT 1 // enable daily limit.
#define USER_GPIO_IN 1 // does "1" mean enable or pin number? If it's a pin number, I need this to be GPIO4.
#define REMOTE_MONITORING 0 // disable remote monitoring.
And if possible, I'd like the voltage measurement to be external, not internal, using a resistor divider.

Thanks.