martin-ger/esp_wifi_repeater #551: Compile error in docker build on windows: config_flash.c:394: multiple definition of `user_rf_cal_sector_set'

NordmeyerT, 2025-02-07:
Trying to build using docker following the manual. 
OS is Windows 11
Leads to error
```log
build/app_app.a(config_flash.o): In function `user_rf_cal_sector_set':
/home/esp/esp_wifi_repeater/user/config_flash.c:394: multiple definition of `user_rf_cal_sector_set'
/home/esp/esp_wifi_repeater/../esp-open-sdk/sdk/lib/libmain.a(user_rf_cal_sector_set.o):user_rf_cal_sector_set.c:(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
Makefile:135: recipe for target 'firmware/0x02000.bin' failed
 ```

Removing the function _user_rf_cal_sector_set_ from file _config_flash.c_ solves the issue.

Full console output:
```log
esp@e64254eb300b:~/esp_wifi_repeater$ make
CC driver/spi.c
CC driver/new_uart.c
CC user/config_flash.c
CC user/ringbuf.c
CC user/rboot-ota.c
CC user/acl.c
CC user/user_main.c
CC user/rboot-api.c
CC user/sys_time.c
CC mqtt/mqtt.c
CC mqtt/mqtt_msg.c
CC mqtt/utils.c
CC mqtt/ringbuf_mqtt.c
CC mqtt/proto.c
CC easygpio/easygpio.c
AR build/app_app.a
build/app_app.a(config_flash.o): In function `user_rf_cal_sector_set':
/home/esp/esp_wifi_repeater/user/config_flash.c:394: multiple definition of `user_rf_cal_sector_set'
/home/esp/esp_wifi_repeater/../esp-open-sdk/sdk/lib/libmain.a(user_rf_cal_sector_set.o):user_rf_cal_sector_set.c:(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
Makefile:135: recipe for target 'firmware/0x02000.bin' failed
make: *** [firmware/0x02000.bin] Error 1
esp@e64254eb300b:~/esp_wifi_repeater$ ^C
esp@e64254eb300b:~/esp_wifi_repeater$ ^C
esp@e64254eb300b:~/esp_wifi_repeater$ ^C
esp@e64254eb300b:~/esp_wifi_repeater$ ^C
esp@e64254eb300b:~/esp_wifi_repeater$ ^C
esp@e64254eb300b:~/esp_wifi_repeater$ ^C
esp@e64254eb300b:~/esp_wifi_repeater$ make
build/app_app.a(config_flash.o): In function `user_rf_cal_sector_set':
/home/esp/esp_wifi_repeater/user/config_flash.c:394: multiple definition of `user_rf_cal_sector_set'
/home/esp/esp_wifi_repeater/../esp-open-sdk/sdk/lib/libmain.a(user_rf_cal_sector_set.o):user_rf_cal_sector_set.c:(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
Makefile:135: recipe for target 'firmware/0x02000.bin' failed
make: *** [firmware/0x02000.bin] Error 1
```

martin-ger, 2025-02-08:
Should be fixed with the latest commits.

hmueller01, 2025-02-08:
Can you please verify this with #550 
You use the docker environment, right?

NordmeyerT, 2025-02-11:
Yes, this works now, thanks