martin-ger/esp_wifi_repeater #552: docker based "make flash" fails on Windows

NordmeyerT, 2025-02-07:
On Windows 11
Calling make flash" fails with following error

```log
esp@e64254eb300b:~/esp_wifi_repeater$ make flash
/home/esp/esp_wifi_repeater/../esp-open-sdk/xtensa-lx106-elf/bin/esptool.py --port /dev/ttyUSB0 --baud 460800  write_flash -ff 40m -fm dio -fs 4MB 0x00000 firmware/0x00000.bin 0x02000 firmware/0x02000.bin 0x82000 firmware/0x82000.bin 0x3FC000 /home/esp/esp_wifi_repeater/../esp-open-sdk/sdk/bin/esp_init_data_default_v08_vdd33.bin
usage: esptool write_flash [-h] [--flash_freq {40m,26m,20m,80m}]
                           [--flash_mode {qio,qout,dio,dout}]
                           [--flash_size {detect,4m,2m,8m,16m,32m,16m-c1,32m-c1,32m-c2}]
                           [--no-progress] [--verify]
                           <address> <filename> [<address> <filename> ...]

```

hmueller01, 2025-02-08:
This is fixed with #550  ...

NordmeyerT, 2025-02-11:
Now I get 

```log
esp@cc1e02be70ae:~/esp_wifi_repeater$ make flash
/home/esp/esp_wifi_repeater/../esp-open-sdk/xtensa-lx106-elf/bin/esptool.py --port /dev/ttyUSB0 --baud 460800  write_flash -ff 40m -fm dio -fs 4MB 0x00000 firmware/0x00000.bin 0x02000 firmware/0x02000.bin 0x82000 firmware/0x82000.bin
usage: esptool write_flash [-h] [--flash_freq {40m,26m,20m,80m}]
                           [--flash_mode {qio,qout,dio,dout}]
                           [--flash_size {detect,4m,2m,8m,16m,32m,16m-c1,32m-c1,32m-c2}]
                           [--no-progress] [--verify]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument --flash_size/-fs: invalid choice: '4mb' (choose from 'detect', '4m', '2m', '8m', '16m', '32m', '16m-c1', '32m-c1', '32m-c2')
Makefile:161: recipe for target 'flash' failed
make: *** [flash] Error 2
```

NordmeyerT, 2025-02-11:
this fixes the issue in Makefile

```log
-ESPTOOLOPTS    = -ff 40m -fm dio -fs 4MB
+ESPTOOLOPTS    = -ff 40m -fm dio -fs 4m
```

hmueller01, 2025-02-11:
It was 32mb before. But this is deprecated. Modern esptool.py versions use

Flash Size: --flash_size, -fs
Size of the SPI flash, given in megabytes.

Valid values are: keep, detect, 256KB, 512KB, 1MB, 2MB, 4MB, 2MB-c1, 4MB-c1, 8MB, 16MB

See https://docs.espressif.com/projects/esptool/en/latest/esp8266/esptool/flash-modes.html

So this is again a problem of the old toolchain in the docker ... so we should revert this.

hmueller01, 2025-02-11:
As I said, the version in the docker is VERY old ...
```
/home/esp/esp_wifi_repeater/../esp-open-sdk/xtensa-lx106-elf/bin/esptool.py version
esptool.py v1.2

```

My SDK comes with esptool.py v2.8 resulting in 
```
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.

```
Modern environment (PlatformIO):
```
~/.platformio/packages/tool-esptoolpy/esptool.py version
esptool.py v4.5.1

```
only supports
`--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}
`
`esptool write_flash: error: argument --flash_size/-fs: invalid choice: '32m' (choose from detect, keep, 256KB, 512KB, 1MB, 2MB, 2MB-c1, 4MB, 4MB-c1, 8MB, 16MB, 32MB, 64MB, 128MB)
`

> this fixes the issue in Makefile
> 
> ```
> -ESPTOOLOPTS    = -ff 40m -fm dio -fs 4MB
> +ESPTOOLOPTS    = -ff 40m -fm dio -fs 4m
> ```
It's megabits, so 4m == 512kB ... not good.

hmueller01, 2025-02-11:
reverted, see #554