martin-ger/esp_wifi_repeater #557: wifi scan

FreeCode911, 2025-03-16:
can we just add a button to scan wifi and see the wifi name , wifi dBm , 
or any one do it plz let me know

developersakibur, 2025-04-05:
I came across a .bin file for the ESP8266 that has the Wi-Fi scan feature enabled, including RSSI (dBm) values. It also allows saving the scan results either temporarily or permanently.

Would it be possible for you to help integrate this functionality into the repository? I believe it would be a valuable addition.

Looking forward to your response.

[image]
[image]

FreeCode911, 2025-04-07:
oh